far-far-away-science/hab-v2 #1: List APRS messages formats we should be using

usaguerrilla, 2016-04-01:
- potentially with 2 GPS (which ID do we change -11?)
- we need to send this (no need to send all in one message apart from time/GPS coordinate/altitude/number of satellites:
  - time
  - GPS coordinates 
  - Altitude
  - Heading
  - Speed
  - Battery voltage
  - Number of satellites
  - Message #
  - CPU temperature (need to research this)
  - Batteries temperature
  - Max acceleration experienced X,Y,Z

Might think about more stuff.

N6UDP, 2016-04-26:
Ok, so the best option is to do a separate telemetry packet after the position one.. example:
2016-04-25 20:39:31 UTC: VK5RAC-1>APNU19,EYRE2-2,qAR,VK5ZEA-1:!3423.56SS13557.24E#PHG5630/ Eyre Pen. Ui-Digi PF75XO @330m
2016-04-25 21:02:05 UTC: VK5RAC-1>TLM,EYRE2-2,qAR,VK5ZEA-1:T#091,048,138,057,021,001,00000000,SA3VFD
As for a Balloon you want it to be something like 4747.50N/12215.10WOTesting
the O being balloon.
As for the format of the telemetry http://www.eoss.org/hardware/w5vsi_gps/telemetry_gps has a good breakdown.  http://www.aprs.net/vm/DOS/TELEMTRY.HTM is the official thing.

usaguerrilla, 2016-07-02:
done

closed 2016-07-02 by usaguerrilla