far-far-away-science/hab-v2 #13: Add pressure sensors and collect infromation from them

usaguerrilla, 2016-04-01:
Do we want to collect pressure close to the ground?
Do we want pressure from up high?
Pick sensors.
Need to save to SD.
Should we send it over the radio once in a while?

usaguerrilla, 2016-04-04:
http://www.ebay.com/itm/Sensore-pressione-MD-PS002-vacuum-sensor-absolute-pressure-senso-for-Arduino-/191674522193?hash=item2ca0b11a51:g:XIYAAOSwq5lTnsvo

http://www.ebay.com/itm/Freescale-MPXV6115VC6U-vacuum-sensor-115-to-0-kPa-16-7-to-0-psi-USA-seller-/231887983618?hash=item35fd99d402:g:a~AAAOSwzvlW8yGV

usaguerrilla, 2016-04-07:
for this task use one of these sensors (lower altitude ones):

http://www.digikey.com/product-search/en/sensors-transducers/pressure-sensors-transducers/1966259?k=pressure+sensor&pv5=2394&FV=3680002%2Cfff4001e%2Cfff800b3%2C24003de%2C24003df%2C24003e0%2C24003e1&mnonly=0&newproducts=0&ColumnSort=0&page=1&stock=1&quantity=0&ptm=0&fid=0&pageSize=25

usaguerrilla, 2016-07-08:
sensor make and model are in the sensors list in wiki
this is an SPI sensor.

usaguerrilla, 2016-07-08:
Sensors list:
https://github.com/far-far-away-science/hab-v2/wiki/sensors_list

stcarlso, 2017-09-15:
BME280 sensor is implemented in Phoenix

closed 2017-09-15 by stcarlso