far-far-away-science/hab-v2 #21: Add LORA or other radio options to supplement APRS for package recovery

usaguerrilla, 2016-04-01:
Check power requirements.
Turn them on, once package lands.

usaguerrilla, 2016-06-07:
LORA should:

- after we turn on electronics on the ground LORA should power up and start transmitting for 5 minutes (every 15 seconds or so, duration and delay can be discussed). We want to make sure it works.
- after this it should power down
- when altitude is < 3 km LORA should again power up (as range is something like 10km) we might receive messages from it in case something is wrong with APRS and vice versa and transmit every 30 seconds or so
- when on the ground for 1st 3 hours it should keep transmitting every 15-30 seconds or so (check how much power will it consume)
- after 3 hours it should transmit every 15-30 minutes

to discuss: do we need to handle case where it landed one a moving object (river, truck) and do we need to transmit more often in this case?

stcarlso, 2017-09-15:
Implemented in Phoenix, LoRA is active at all times and transmits with APRS.

closed 2017-09-15 by stcarlso