far-far-away-science/hab-v2 #24: Migrate Copernicus GPS code to STM32

usaguerrilla, 2016-04-03:


usaguerrilla, 2016-04-06:
requirements:
* code should support recovery from temporary loss of power or reset and loading last known position to transmit over APRS (with flag of stale information).
* new position should be transmitted as soon as GPS acquires lock
* this is the 2nd highest priority after APRS signal modulation
* GPS should not be updated more than once a second

recovery data requirements:
* recovery data will be stored in EEPROM in both banks
* each bank will have ring buffer for 50 records (after we reach end we start overwriting old records). 50 was picked to prolong EEPROM lifetime.
* on reboot we will scan all records (from banks which are online, if bank is offline for writing wait for read operation to be completed before restoring the data) and use latest one
* banks will be written in round robin style (bank 1, bank 2, bank 1, etc.)

recovery record spec:
* TODO

open items:
* what will happen if during EEPROM write operation MCU reboots or has fault ISR or power off event
* how to determine buffer writing round robin order

usaguerrilla, 2016-04-06:
Unlock with wait until write is completed:
/* (1) Wait till no operation is on going */
/* (2) Check if the PELOCK is unlocked */
/* (3) Perform unlock sequence */
while ((FLASH->SR & FLASH_SR_BSY) != 0) /* (1) */
{
/* For robust implementation, add here time-out management */
}
if ((FLASH->PECR & FLASH_PECR_PELOCK) != 0) /* (2) */
{
FLASH->PEKEYR = FLASH_PEKEY1; /* (3) */
FLASH->PEKEYR = FLASH_PEKEY2;
}

usaguerrilla, 2016-04-06:
Lock with wait until write is completed:

```
/* (1) Wait till no operation is on going */
/* (2) Locks the NVM by setting PELOCK in PECR */
while ((FLASH->SR & FLASH_SR_BSY) != 0) /* (1) */
{
/* For robust implementation, add here time-out management */
}
FLASH->PECR |= FLASH_PECR_PELOCK; /* (2) */
```

usaguerrilla, 2016-04-06:
Write to data EEPROM code example
```
*(uint8_t *)(DATA_E2_ADDR+i) = DATA_BYTE;
*(uint16_t *)(DATA_E2_ADDR+j) = DATA_16B_WORD;
*(uint32_t *)(DATA_E2_ADDR) = DATA_32B_WORD;
```
DATA_E2_ADDR is an aligned address in the data EEPROM area.
i can be any integer.
j must be an even integer.

usaguerrilla, 2016-04-06:
Erase to data EEPROM code example
/* (1) Set the ERASE and DATA bits in the FLASH_PECR register
to enable page erasing */
/* (2) Write a 32-bit word value at the desired address
to start the erase sequence */
/* (3) Enter in wait for interrupt. The EOP check is done in the Flash ISR
*/
/* (6) Reset the ERASE and DATA bits in the FLASH_PECR register
to disable the page erase */
FLASH->PECR |= FLASH_PECR_ERASE | FLASH_PECR_DATA; /* (1) */
*(__IO uint32_t *)addr = (uint32_t)0; /* (2) */
__WFI(); /* (3) */
FLASH->PECR &= ~(FLASH_PECR_ERASE | FLASH_PECR_DATA); /* (4) */

Scorillo47, 2016-04-06:
My suggestion would be something along these lines (pseudo-code only - I didn’t add CRC, sequence#, Flash synchronization, etc). Sorry for the Outlook-enforced capitalization: 

Bool Write(data) 
{
                For(int = 0; I < maxAddressesToBeTried; i++)
                {
                                addr = PickUpNextLocation(); // alternate banks round-robin, etc.
                                For(int = 0; I < maxRetries; i++)
                                {
                                                If (Write(data, addr) == success)
                                                {
                                                                // wait a bit … 

                                                                // Read after write to ensure the correct data is there
                                                                Data1 = Read(addr);
                                                                If (Data == Data1)
                                                                {
                                                                                Return true;
                                                                }
                                                }

                                                // Writing failed (bad block?)

                                                // wait a bit more …                                        

                                                // retry writing on the same location maxRetries
                                }
                                // give up on this address. Try writing on the next location (up to maxAddressesToBeTried tries)
                }

                Return false;
}

usaguerrilla, 2016-04-07:
algorithm in a nutshell:

```
isDataValid(address)
{
    return true if crc(from sequence id and payload at this address) == crc(crc at this address) else false
}

recoverDataFromBank(bankAddress, currentMaxSequenceId)
{
    if (!waitUntilBankIsAvailable(TIMEOUT))
    {
        return (none, none)
    }
    data = none
    maxSequenceId = currentMaxSequenceId
    foreach (address in bankAddress)
    {
        if (isDataValid(address) && getId(address) > maxSequenceId)
        {
            data = getData(address)
            maxSequenceId = getId(address)
        }
    }
    return (data, maxSequenceId)
}

recoverGpsData(data)
{
    sequence_id maxGoodId;
    (data, sequence_id) = recoverDataFromBank(Bank1Address, none)
    (data, sequence_id) = recoverDataFromBank(Bank2Address, sequence_id)
    return data != none
}
```

usaguerrilla, 2016-04-07:
update recovery algorithm in a nutshell:

```
updateRecoveryData(data)
{
}
```

usaguerrilla, 2016-04-07:
This algorithm protects against the following problems:

* MCU resets (or experiences short power interruption) and we cannot communicate with GPS after power on
* GPS unit looses last known good position or overwrites it with the empty position message

* This algorithm / hardware design is:
* (2*N-2)x fail operational / fail safe for EEPROM individual memory cells failures (couldn't read or couldn't write at specific addresses).
* fail safe for EEPROM data banks (as we have 2 and failure of one bank will still leave system running).
* fail safe for GPS loosing last known good position

N - size of a ring buffer in one EEPROM memory bank

usaguerrilla, 2016-05-05:
code migrated. gps backup will be done by #37

closed 2016-05-05 by usaguerrilla